symbench/scm #2: Overall architecture

brollb, 2021-05-03:
This is an issue to have a discussion about the overall project architecture. Currently, I think we should prioritize writing small, modular components rather than a large IDE-like experience. This will make it easier to cherry pick components to use as well as resuse in a bigger system like the electric circuits design studio.

Here are the different components that might be a good starting place:
- scraper: CLI-based tool for fetching diagrams from various sources.
- labeler: browser-based labeling tool for annotating images. It would be nice to give interactive feedback to the user about the implications of the annotations (ie, what other components are now recognized in the image?)
- circuit recognizer/maker (CLI). Once we have annotations that we think are sufficient, we should be able to generate them using a standard CLI workflow. I am imagining something like `scm.recognize -a annotations.json --source annotated-image.png <list of files>`
- circuit maker (REST API). This will probably be required for the labeler but should be pretty minimal.

umesh-timalsina, 2021-05-03:
> Currently, I think we should prioritize writing small, modular components rather than a large IDE-like experience. This will make it easier to cherry pick components to use as well as resuse in a bigger system like the electric circuits design studio.

I totally agree. There should be a lot of moving pieces to this repository and those pieces and we should be able to easily integrate these pieces to integrate them to other code bases.

> scraper: CLI-based tool for fetching diagrams from various sources.

scrapy-cli uses `optparse` under the hood. This is great for us and some custom command line arguments are already supported. We can may be further customize the scrapy commands to fit our needs.

> labeler: browser-based labeling tool for annotating images. It would be nice to give interactive feedback to the user about the implications of the annotations (ie, what other components are now recognized in the image?)

As was decided, this should be a modular component (at this point `svelte` is the best option).

> circuit recognizer/maker (CLI). Once we have annotations that we think are sufficient, we should be able to generate them using a standard CLI workflow. I am imagining something like scm.recognize -a annotations.json --source annotated-image.png <list of files>

This should probably be another `cli subcommand` module. `scm.scrape` for scraping and `scm.recognize` for annotations/recognition.

> circuit maker (REST API). This will probably be required for the labeler but should be pretty minimal.
What would be the purpose of this REST API? I would imagine for programitacally annotating the images?

brollb, 2021-05-03:
> What would be the purpose of this REST API? I would imagine for programitacally annotating the images?

I was thinking that the labeler should probably give immediate feedback to the user about the additional components that are now recognized after adding the annotation. If this is the case, it would need to have some way to use the "recognize" command (hence the REST API).

umesh-timalsina, 2021-05-31:
This probably is the best issue to track this, so I am adding it here:

What's lacking in the current architecture (Non-Exhaustive List):
1. Uses Component and Port Names and Annotation Colors from a JSON file rather than requesting server.
2. With #25, port detection workflow has been added but it uses example json file rather than server capabilities. See the workflow checklist in #25, [here](https://github.com/symbench/scm/pull/25#issue-657946863).

For enhancements, we need a a well defined REST API with the following capabilities:
1. Endpoints that get rid of the `json` files from the front end.
2. Annotation Pipeline endpoints with port detection capbilities.

This is just a suggestion, but I think it will get most of the issues resolved.