symbench/scm #4: Recognize to Annotator API

umesh-timalsina, 2021-05-06:
It might be nice to rapidly prototype a python API using FastAPI. The current UI is functional for annotating but its not very helpful without the `recognize` part of this project.

brollb, 2021-05-06:
Good find! I just saw this. I have put together most of a flask implementation but it looks like it would be pretty trivial to migrate over.

As the complexity was starting to grow a little bit, I recently refactored the recognize script to an actual python package. (I am not married to the name so we can change it to something like `scm.recognize` if you want.) Anyway, I started on an API using flask [here](https://github.com/symbench/scm/blob/main/circuit_recognizer/circuit_recognizer/server.py) that could be migrated to use FastAPI instead if you want.

umesh-timalsina, 2021-05-10:
Base API Implementation is handled using FastAPI. Closing this.

closed 2021-05-10 by umesh-timalsina