symbench/scm #51: Disable tag validation for now?

brollb, 2021-06-08:
It might be nice to disable tag validation as we don't yet know how we will handle complex things like subcircuits or varying levels of abstraction. If validation is disabled, we can experiment more easily with how this should be handled. (It should probably be re-enabled later once we figure out how exactly we would like to handle subcircuits and abstract components.)

umesh-timalsina, 2021-06-08:
Okay. We can probably only limit the checks to the following:
1. It should have a single element body with length 1.

umesh-timalsina, 2021-06-08:
This would also mean we need to add some styles for `AgentCreated` section for components with no predefiend tags.

commit 79ab613c (2021-06-09): Relax validation for annotations. Closes #51 (#57)

* Relax validation for annotations. Closes #51

* WIP- Every annotation is valid now
closed 2021-06-09 by umesh-timalsina