symbench/scm #56: IndexError in recognize_ports

umesh-timalsina, 2021-06-09:
```py
    return await dependant.call(**values)
  File "/home/umesh/work/design.R/scm/circuit_recognizer/circuit_recognizer/server.py", line 36, in detect_ports
    ports = recognize_ports(image, annotation)
  File "/home/umesh/work/design.R/scm/circuit_recognizer/circuit_recognizer/recognition.py", line 81, in recognize_ports
    r_line.extend(image, [annotation])
  File "/home/umesh/work/design.R/scm/circuit_recognizer/circuit_recognizer/connect/line.py", line 160, in extend
    while image[leader[0], leader[1]]:
IndexError: index 272 is out of bounds for axis 0 with size 272

```

[image]

brollb, 2021-06-09:
Can you share the annotations?

umesh-timalsina, 2021-06-09:
[annotations.zip](https://github.com/symbench/scm/files/6626505/annotations-sample_circuit_id-2021-06-09T19_43_09.266Z.zip)