symbench/scm #70: OpenSeaDragonViewer should be more configurable

umesh-timalsina, 2021-06-10:
Currently, following configurations are possible via query strings:
1. `ImageURL`: Ability To load Custom image
2. `apiURL`: For template matching and ports matching

Following would be nice to have:
1. Template Root matching for existing annotations
2. Load Multiple Images via query strings
3. Load Zip File of annotations and images.  etc... 
This is a rough thought based on a discussion with @brollb.

brollb, 2021-06-10:
Here are my thoughts:
- I like the [current code structure](https://github.com/symbench/scm/blob/main/js/circuit-annotator/src/App.svelte#L2-L15) as it ensures there is both a URL query string parameter for the setting as well as a programmatic API.
- As far as the API goes, it seems like we want to be able to support the following capabilities:
    - Opening custom image(s) (perhaps we should change the qs param to "images=<comma separated list>")
    - Customizing the API endpoint to use for template matching, etc
    - Opening existing annotated image
    - Opening a list of annotations to use for component detection (from previous annotated images)

I think we can probably just support them as follows:
- `images=<comma separated list or a single image>`. Maybe we should support both `image` and `images`... It isn't a big deal either way but might be a little annoying if a constructed URL doesn't behave as expected (and they might be easy to mix up).
- `api=<api url>`
- `annotations=<comma separated list to json files>`. If there is no image specified, we should infer the images from the annotations (ie, all unique `target.source`s of the annotations). If images are set, then the annotations should be shown if they match a displayed image URL. Otherwise, they should just be used for the recognition API.

Thoughts?

umesh-timalsina, 2021-06-10:
> Opening custom image(s) (perhaps we should change the qs param to "images=")

- I like this approach and may be we should stick to a single query param for loading images. `images` can be a comma spearated list of ImageURLs that we can provide. Upon loading of multiple images, we should be able to open the viewer in collection mode (until multi-image support comes to the viewer :) ). 

> api=<api url>

 This exists currently. However, its up to the viewer to decide which endpoints to query for ports and which for components. which I think is reasonable., 

> annotations=<comma separated list to json files>. If there is no image specified, we should infer the images from the annotations (ie, all unique target.sources of the annotations). If images are [...]

agreed. 

On a further note:

May be we draft up a template annotation to use for some circuit images that cover most components and upon loading of a new image to the viewer, we use that annotation to do template matching

Quick Sidenote
May be we also decide on the precedence of these query parameters in a way such that mixing them results in a clear predefined overriding behavior.

commit ac7bcba9 (2021-06-30): Additional Configuration Options for OSD Viewer via Query params. Closes #70 (#78)

* WIP- #70 Add SequenceMode for OSDViewer in multiImageMode

* WIP- Merge branch master and support OSD Pagination plugin

* Emitter support for pagination plugin

* WIP- emitter is now an optional argument

* WIP- annotorious-osd released

* WIP- Address PR review comments

* WIP- Trim and filter queryParams

* WIP- Remove unused variable, fix missing semicolon

* WIP- Fix eventlistener performance issues
closed 2021-06-30 by umesh-timalsina