symbench/scm #121: Port outside the component

brollb, 2021-07-01:
It appears that ports can be outside the parent by a pixel. I suspect this is due to rounding errors:
[image]

umesh-timalsina, 2021-07-16:
Another instance of the issue:
[image]

umesh-timalsina, 2021-09-23:
Is this fixed @brollb? I no longer encounter this.

brollb, 2021-09-23:
I haven't seen this in a while either so I suspect it might be. (I don't recall for sure.) We can probably go ahead and close it and revisit it if it occurs again.

closed 2021-09-23 by brollb