symbench/scm #160: Documenting annotation scenarios

umesh-timalsina, 2021-07-09:
## Documenting Annotation Scenarios
Okay, there are multiple events and with the changes in the #152, we have an annotation workflow. However, the workflow in #152 might not be enough. In this issue, I am maintaining a running list of all the scenarios that come to mind and add additional opinions and additions to this document. This can live as an issue for now but, over time this can be maintained as a road-map/ annotation document as well.

Before elucidating the scenarios, lets look into the types of annotations:

|Type            | Origin |
|-----------------|-----------|
| Component Annotation | Agent|
| Port Annotation | User  Created|
| Component Annotation | User Created|
| Port Annotation | Agent|

We apply different styles to different components and ports inherit the component styles. User Created and Agent generated annotations have solid vs. dashed bounding boxes. These are the shown below:
[image]

[image]

### Scenarios
Below, I list the following scenarios based on what came to me while writing this up. This list is certainly non-exhaustive:

-  User Creates Component with a label :  Upon user creating annotation and labeling a component, we request the server for the ports of that component. All the ports coming from the server/agent will be tagged `Port`.
- User updates ports for  User created component: If user updates ports for user created annotations and all of its ports are fully labeled, a `REQUEST_TEMPLATE_MATCH` request is sent to the server, which is implemented in #152. This request will take the annotation snippet, and upon success, all the matched annotations are added to the viewer.
- User moves the annotation target: Upon moving the annotation target, underlying ports are deleted and upon update, the ports are re-requested. Note: the ports will have the same labels and at this point, users will have to manually label them. This is certainly a scenario that I think we can improve on.
- User deletes an annotation: If user deletes an annotation, there are three scenarios:
    * The annotation is a Port Annotation: Simple Deletion Happens
    * The annotation is a Component Annotation that was agent generated: The annotation and its ports are deleted
    * The annotation is a Component Annotation that was user generated: If any other canonical forms of that annotation exist in the viewer, a prompt is shown to the user if they want to delete the canonical forms the annotations. If confirmed, all the canonical matches and their ports are deleted, else only the current component and its ports are deleted.

### Interlude: Viewer Modes
Currently we support the following viewer modes:
1. Single Image: A single image in the viewer
2. Multi-Image Sequence: Multiple Images are open in the viewer but only one image is shown at a time
3. Multi-Image collection: Multiple Images are open in the viewer and all of them are shown at once

All the scenarios mentioned above should work as described for the given modes, unless we have some bugs in the code base :slightly_smiling_face:.

### Further thoughts and unaddressed scenarios:
* A manual port annotation workflow has been discussed in #131 but not yet implemented.
* An ability to filter annotation by component type, which will be pretty helpful if we are to implemented, which might be helpful for Multi-Image collection mode added in #116. 

_Notes: We need more thoughts and work on this._

brollb, 2021-07-09:
I think this makes sense. A couple minor comments:
- We shouldn't refer to the current implementation used for component detection from the server (ie, template matching) in the request name since this could change in the future. We should probably use a more general term like `REQUEST_DETECT_COMPONENTS` or something.
- I think we should really prioritize labeling more of the dataset by hand and using that experience to prioritize issues like #131. So far, I haven't found myself needing that feature so I suspect it isn't a high priority but perhaps this isn't the same for others.

umesh-timalsina, 2021-07-09:
> I think this makes sense. A couple minor comments:
>
> * We shouldn't refer to the current implementation used for component detection from the server (ie, template matching) in the request name since [...]

This request is still in a  PR, I will change it to `REQUEST_DETECT_COMPONENTS`

> * I think we should really prioritize labeling more of the dataset by hand and using that experience to prioritize issues like [Manual Port Annotation Workflow #131](https://github.com/symbench/scm/issues/131). So far, I haven't found myself needing that feature so I suspect it isn't a high priority but perhaps this isn't the same for others.

So far this hasn't been an issue for me, but I suspect this might be needed where the agent/server is not successful in Port detection.