symbench/scm #174: Undo agent-created annotations?

brollb, 2021-07-12:
We need to have some way to recover when the annotation is subtle and results in many detected components. See the effect of labeling a junction below:
[image]

umesh-timalsina, 2021-07-12:
Couple of question regarding this:
1. Should the undo only effect agent generated annotations?
2. Should there be a redo?

brollb, 2021-07-12:
It might be worth adding an option to add a singular annotation. Or maybe template matching sensitivity should be handled first.

My thinking is that the motivation for undo here is bc it selected way to many things that I cannot recover from. Currently, I would need to go and delete all the false positives. In other words, there are two problems:
- How can I efficiently annotate all the junctions?
- How can I recover from this list of way too many annotations?

I think we should be clear about these bc updating the first one might change our view of the questions you mentioned above. My initial thoughts are:
1. Yes. In my case, the original annotation was fine but the inferred ones were largely incorrect. If we had the ability to control the sensitivity of the matching or something, this might let me redo this.
2. Probably.

umesh-timalsina, 2021-07-12:
> How can I recover from this list of way too many annotations?

My suggestion for handing this would be to implement undo/redo and implement a toolbox (GMIP comes to mind). This tool box will start simple: [Annotate, Delete, Copy(optional) etc..]. Undo redo will provide a coarse control and the toolbox can be used to provide a fine-grain control. 

> How can I efficiently annotate all the junctions?
My idea was to not request ports by adding a validator for PortAnnotator plugin i,e. don't request port for Junctions.

brollb, 2021-07-12:
> My suggestion for handing this would be to implement undo/redo and implement a toolbox (GMIP comes to mind). This tool box will start simple: [Annotate, Delete, Copy(optional) etc..]. Undo redo will provide a coarse control and the toolbox can be used to provide a fine-grain control.

This sounds like it is answering the question of how to annotate the junctions, isn't it? Or are you implying that we could use something like "rectangle select" to easily delete the incorrect annotations? The two sub-problems were meant to basically be: 
1) How can we prevent something like this from going wrong? 
2) How can we recover when something like this goes wrong? 

Maybe I am missing something but isn't your answer answering the second one rather than the first? Overall, I am all for a toolbox if we have rich enough features to support it but was a bit puzzled about how it connected to the quoted comment.

> My idea was to not request ports by adding a validator for PortAnnotator plugin i,e. don't request port for Junctions.

Ports are actually no longer requested as of https://github.com/symbench/scm/pull/172

umesh-timalsina, 2021-07-12:
> This sounds like it is answering the question of how to annotate the junctions, isn't it? Or are you implying that we could use something like "rectangle select" to easily delete the incorrect annotations? The two sub-problems were meant to basically be:

> 1. How can we prevent something like this from going wrong?

I think this has to be done from server side or can it be handled from the front-end?

> 2. How can we recover when something like this goes wrong?

The answer provided [above](https://github.com/symbench/scm/issues/174#issuecomment-878546876) might be specifically related to this.

brollb, 2021-09-23:
As of #235, the sensitivity of the component detection can be explicitly adjusted. Turning it all the way up will basically remove all matches and allow the user to recover from poor annotations. This kinda adds support for the first point, too, as it allows the user to turn the sensitivity up or down to get the exact annotations in question. (That said, the slider could probably be improved as most matches are very close to one another...)

closed 2021-09-23 by brollb