symbench/scm #183: Labeling SubCircuits

umesh-timalsina, 2021-07-19:
We need to figure out a strategy for labeling subcircuits. Simple example below for `fc.png`.
[image]

_Originally posted by @umesh-timalsina in https://github.com/symbench/scm/issues/169#issuecomment-882633430_

umesh-timalsina, 2021-07-19:
Possible duplicate of #175.

brollb, 2021-07-28:
One approach that would be good to try would be:
- Label the entire region as a "Circuit"
- Label the individual ports using "Port" as the label. Set the comment text to the name of the port.

My hope is that the segmentation model might be able to learn how to segment out the subcircuits. Currently, I am also training a separate model for detecting the ports. After detecting the ports, we could use OCR to hopefully determine the port names.

brollb, 2021-07-28:
This workflow is currently blocked by #131

brollb, 2021-09-23:
Currently, the plan is to label them as described above. In other words, the entire circuit is labeled as "Circuit" and the individual ports are "Port" with the text recorded in the comment.

closed 2021-09-23 by brollb