symbench/scm #206: Component Detection Sensitivity

brollb, 2021-08-02:
The sensitivity for component detection is reasonable for most diagrams but sometimes it is not nearly sensitive enough. See the figure below:
[image]

This raises a couple questions:
- Should we be able to delete only the detected matches (not the original)?
- What if users could have a slider to interactively see all the matches?

brollb, 2021-08-16:
This slider could be useful: 

Currently, my thoughts are to:
- Change the [minimum threshold](https://github.com/symbench/scm/blob/main/circuit_recognizer/circuit_recognizer/recognition.py#L53) for the matches that have been returned
- Include the similarity score in the returned results (as part of the annotation?)
- Add a slider for automatically hiding/showing annotations around a given range. (Maybe using [this library](https://svelte.dev/repl/e93d43c42b434e3aa3e2a5815f805c75?version=3.22.3)?)

It would be nice if we could add this as another button on the widget used when creating annotations so we could have options like: "cancel", "ok", "detect matches". In this case, "ok" would be changed to only label the current one whereas "detect matches" would provide a slider to allow us to select a bunch of matches. Unfortunately, when I have been looking at the library, it doesn't appear that adding new buttons is possible. @umesh-timalsina, have you seen anything promising on this front?

An alternative approach could be to add the slider to the widget itself and have it only enabled for existing annotations (with all the ports). Otherwise, it could be disabled with help/hover text specifying what is required before it can be used. Hmm... This actually might not be a bad approach to go in general :)

umesh-timalsina, 2021-08-17:
I think the widgets we add are generally appended into their editor component which has the buttons, I haven't seen anybody actually changing the editor, but should be possible with some customization. At least this was the state of the tool last time I saw it. Another alternative could be (and a better one IMO).
-  Manual removal button for detected matches based on your slider approach, but this button would live in the viewer.

In other words, user could use this button to change the slider values and the matched annotations are cleared or added depending on the current slider value. In the diagram above, all the matches for a selected annotation will be added/removed based on the slider movements in the viewer.

brollb, 2021-08-17:
My concern with that approach is that it seems to assume that there is a single "correct" value for the slider whereas the user will likely want to change it based on the component. Even when I am labeling components now, I don't actually have issues with the sensitivity at a diagram level (ie, for all the components in a diagram) but with specific components that often match other artifacts in the figure such as junctions, ground, and capacitors.

umesh-timalsina, 2021-08-17:
That is true, however, if you tie that to a selected annotation, I think we are good.

commit b92fdff0 (2021-09-23): Customize component sensitivity. Closes #206 (#235)

* Add confidence to creator annotation

* Add detection sensitivity widget

* Add annotation filter for hiding annotations given the sensitivity

* Disable sensitivity for agent-created annotations

* Update components appropriately wrt sensitivity

* Update annotation filter

* Remove derived components on component identification

* Remove derived annotations before recognition

* Remove debug comment, EOL
closed 2021-09-23 by brollb