symbench/scm #239: OCR Helper Widget for PortAnnotation Mode

umesh-timalsina, 2021-09-22:
For named ports in subcircuits, it is annoying to having type the name of the port for every annotation we create. It might be nice to incorporate a simple OCR helper when in port annotation mode that tries to recognize the characters of a named port. While the functionality seems primitive, we need to jump quite a few hoops to accomplish this, listed below:

1. Swap the comment widget (default) with another widget 
2. Upon `createSelection` in port annotation mode, select the image snippet and use a plugin to set widget text (based on OCR results).

brollb, 2021-09-22:
Here is a screenshot of what @umesh-timalsina is describing:
[image]

I think this might only be needed for Circuit ports (ie, `Port` types). When we label text, this could be useful there, too. Can you think of any other cases, @umesh-timalsina?

umesh-timalsina, 2021-09-22:
Yeah, I have added the feature keeping that in mind. For now, starting out simple with `pytessaract` (i.e. tesseract)'s default settings, which work well for selected snippets.

commit 0f4fa611 (2021-11-08): Detect text for default attribute values. Closes #239 (#240)

* WIP- #239 selectedSnippetOCRPlugin

* WIP- #239: Server changes

* WIP- #239. Minor code changes for client

* WIP- Fix environment-dev.yml for pytesseract

* WIP- Pad snippet for better results

* Raise HTTPError if tesseract OCR not found in server

* Robust OCR pipeline; testing

* Fix typo in test_recognize

* Add OCR addition criteria based on fragmentMatch

* WIP- don't instantiate error class

* ESLINT fix in plugin

* WIP- Server side support for annotation purpose while performing OCR.

* WIP- Use threshold and annotation [... cut]
closed 2021-11-08 by brollb