symbench/scm #269: Annotating Attributes, Circuits, and Abstract Components

brollb, 2021-10-04:
This issue is not about a technical feature but rather the correct way to represent some of the trickier concepts via annotations. These include:
- attributes
- circuits (and their label text)
- abstract components (labels)
- named ports
- labeled connections

brollb, 2021-10-04:
Here is an example of a labeled connection (from [here](https://github.com/symbench/scm/blob/main/dataset/Images/full/bq24071-bq24070.jpg)):
[image]

umesh-timalsina, 2021-10-04:
Currently, we use the annotation mode, which supports two distinct types of modes:
1. ComponentAnnotationMode: Top level annotations for circuit/components etc..
2. PortAnnotaionMode: Child annotations for Ports (linked by extending the W3C annotations spec by incorporating a `parent_id`)

Clearly these two modes will not be enough for us. @brollb and myself have been discussing the following scenarios (Most of them will be addressed by extending the PortAnnotationMode to be a child annotation mode. On how best to handle these, for me the best bet is to include this information via a dictionary like data structure in `Components.json` and extend the `PortModeToggler` to use that info. This information should contain the following:

1. What are the annotation modes where this component can be used as a parent?
2. Can it be at the top level?
3. What are the different annotation bodies allowed for its child annotation and itself? 

If this information explodes, we can have a separate file to store them. However this is a nice/generic way to handle this information.

umesh-timalsina, 2021-10-04:
> A couple remaining questions/comments:
> 
> * Should we also capture the "raw text" of the content? This would be useful in evaluating the OCR.

> * If we capture, the raw text as well we are essentially collecting labels for both the intermediate representation and the final representation. That is, the raw text can be used to evaluate the segmentation & OCR NNs while the semantic form will be used to evaluate the heuristics that interpret the IR and generate the find version.

Yeah we should. Some heuristics and  KD-Tree search can be exploited to consolidate linked annotations that should be a single one. For example, in the above case, The `200mV` belongs to the Voltage source. How we do that? Search for nearest neighbor and see any of its annotation (tag) values if matches the expected for it, join them

brollb, 2021-10-04:
Labeling named ports is tricky. I think we can break them down into three different categories:
### Named Ports with Visual Indicators/Symbols
The below example (bq24072t-bq24075t-bq24079t) has squares to designate the port with a nearby text attribute for the port name. It also contains a number for the port.
[image]

In this case, I think we should select the port symbol (ie, the rectangle) for the port and treat the others as associated attributes.

Also, it seems our ports may have both `name` and `number` attributes.

### Named Ports w/o Any Symbols
These are ports that are only known to exist bc they have a name next to them. In other words, the connection connects to the component directly.
[image]

For these, I think we should probably include the text in the port annotation and then also label the title attribute. Thoughts?

### Ports as External Entities
I have seen some examples where ports are treated as external components - like an abstract component. An example is below:
[image]

This example is probably not the most important since it is a functional example rather than an example usage of the circuit. Maybe these should be treated like the last one where we select the entire text and edge to treat it as a port...

### Connections Labeled with Port Name
An example of this can be found in the earlier comment about labeled connections. It isn't clear to me what the best way to handle this is. Perhaps we could treat it like named ports (using the arrow heads as the symbols for the port - they are the only ones that seem to have a port name) or we could treat it like named ports without a name.

brollb, 2021-10-04:
## Abstract and Circuit Components
I think the cleanest approach might be to capture the circuit name in the tag details/comment/etc section and then also capture it as an attribute if it is provided. The reason is that some circuit names may only be known from the context (such as in the comment above) and it could still be useful to ensure this info is captured in case it is relevant to any heuristics used later.

Abstract components can be handled the same way except the name will not be quite as clean as a model number and will likely be human-readable instead. An example can be found in the previous comment here, too.

brollb, 2021-10-04:
## Nonstandard Connections
There are a few different representations that are used for connections. For example:
Thick arrows for a bus to some abstract component:
[image]
Connections with text in the center:
[image]
Connections that are not actually connected (see the "C" pseudo-node below):
[image]

umesh-timalsina, 2021-10-05:
> ## Nonstandard Connections
> There are a few different representations that are used for connections. For example: Thick arrows for a bus to some abstract component: [image] Connections with text in [...]

For now we can abstract connections like we abstract components. At this point a direct wire with Manhattan routing seems to be one. The other example be broken but label linked. The other could be connected via an interface like I2C or SPI or even USB etc.,

brollb, 2021-11-09:
Abstract connections should be labeled with a bounding box w/ names as appropriate. Connectivity will be determined using:
- Overlap between the connection and any other components (example 1).
- Standard connection detection (example 2).
- Shared name attributes (example 3).