symbench/scm #304: Annotator refactor to include hide/disable capabilities

brollb, 2021-11-01:
Some of our extensions of the annotator actually affect the `getAnnotations` method. This is not great since currently, the workaround has been to do things like check the port mode and add the parent if needed (ie, it is a leaky abstraction). The annotator should be refactored to include these extensions so it can present a unified abstraction externally.

Specifically, there are currently 2 main features that should probably be handled within the annotator:
- disabling/enabling annotations. These have been called "footprints" in the annotation mode logic.
- hiding/showing annotations. This one could arguably still be handled outside of the annotator if it knows when the image is changed (when in sequence mode) but may be cleaner to still handle within the annotator.

*Note*- When I am saying it should be handled "within the annotator", it doesn't necessarily need to be in the same file but within the same exposed object.

umesh-timalsina, 2021-11-01:
Another piece that could be handled by the annotator is the formatting of annotations and keeping track of existing annotations (Which is currently done in OSD component separately). This would cleanup the `OSD` component a bit as well.

closed 2021-11-01 by brollb

brollb, 2021-11-01:
I merged a PR that was refactoring the features mentioned above. @umesh-timalsina - feel free to open a new PR for the other suggested refactor! (Trying to keep them small :) )

commit f0e85437 (2021-11-01): Refactor disable/hide features in annotator. Closes #304 (#312)

* WIP refactoring annotator w/ features that impact getAnnotations

* Refactor disableable logic to mixin

* Separate UI logic from main logic

* Add hideable and remove filters

* Add tests and clean up types

* Add return type