symbench/scm #361: Testing Complete Annotations

umesh-timalsina, 2021-11-10:
Now that we have been annotating the dataset it might be nice to write some tests to verify we can achieve what we want from the annotations.

brollb, 2021-11-11:
Could you elaborate on "achieve what we want"? Do you mean that we can use the annotated circuit for downstream learning tasks or as a template to import into the circuit design studio?

Or do you mean that the annotations have the structure that we expect - such as labeled ports for all components expecting ports?

umesh-timalsina, 2021-11-11:
In a sense both. We want a good intuition of the dataset we build in a way that our purpose to annotate the circuits is fulfilled and also it can be used to export to circuit design studio. It would mean that we want the following:

1. Consolidation is possible i.e. No errors like https://github.com/symbench/scm/issues/356 should happen for complete a complete annotation.

2. A partial JSON transformation in the format excepted by webgme-json-importer should be possible from annotations. This would mean addition of features either here or in the circuit design studio. I lean towards adding it here because we want a straight forward port from this to a visualizer in electric-circuits.

brollb, 2021-11-11:
To the extent that we can test annotations (or provide feedback from the labeling tool), this is a good idea. Some things will be difficult to test without just implementing the given task we want to be able to support. For me, the goal has been to label enough data to train our segmentation model and verify that the data is sufficient. This is a bit difficult to write automated tests to verify (at least without actually training a model!). When fixing bugs like the one linked above, we should also be adding tests (at least regression tests).

Updating the JSON transformation to the design studio isn't a bad idea to verify that our annotated designs can be imported. It will likely require an update to the metamodel to support concepts like "Abstract".

That said, the way the issue is written, it sounds like there are general tests that you have in mind for the annotations themselves. Is that true? If so, they certainly could make for good tests - or they could be integrated into the labeling tool itself to provide feedback to the user about any outstanding issues. One that comes to mind is if a component is missing its required ports. Did you have anything else in mind?

umesh-timalsina, 2021-11-12:
> That said, the way the issue is written, it sounds like there are general tests that you have in mind for the annotations themselves. Is that true? If so, they certainly could make for good tests - [...]

The way I see it is the following, if an annotation is declared complete
1. It shouldn't encounter issues like #356 (This can be tested)
2. We should be able to consolidate annotations into a hierarchical structure (The hierarchy exists in the client but this support child of child can be extended to incorporate different kind of annotations (not just port) in the server). 
3. The testing could also include things like confirmation to W3C spec.