symbench/scm #362: Split OCR results into correct fields

brollb, 2021-11-11:
It would be great if we could automatically split the OCR results into the correct fields - or at least do a better job of it than it currently does!

Here is what I suggest:
- Since the webgme source model (ie the input to the ontology generation) knows the types of the attributes (text or numeric), we could incorporate that into the ontology as regular expressions.
- When an OCR result is received, each field could select the first match (and remove it) then pass it to the following expected field.

brollb, 2021-11-11:
Just to be clear, I am imagining the regular expressions would be supplied here:
[image]

umesh-timalsina, 2022-01-14:
That seems feasible, couple of comments/ questions:
1. Can the regular expressions here be a comma separated list of regular expressions, just in case if we can't incorporate all the possible representations in a single regex.
2. What do we want to do when there's a match failure for some fields and success for others? Should we just revert back to the current practice of setting everything into value? 
3. Should this be done for manually (i.e. user entered) entered values?

umesh-timalsina, 2022-01-14:
https://www.theiet.org/media/4173/units-and-symbols.pdf This is a comprehensive list of units and symbols used in electrical engineering.

brollb, 2022-01-14:
Good questions. Here are my thoughts:
1. Technically, they should always be able to be expressed in a single regex using `(regex1|regex2|regex3)` etc. It might be easier to read them as a comma separated list. I would defer to whatever is supported by webgme (which looks like only a single regex pattern) since it would be nice not to change the default UI (and they are equivalent so internally, it shouldn't really matter - a single regex is probably easier to support internally).
2. I had been imagining that it would discard any trailing characters that don't match any field. Ideally, we could get user patterns working in tesseract and guarantee that this isn't a problem.
3. I am torn about this. It would be nice but would be problematic when encountering text like `1\Omega5` for 1.5 ohms. Probably not for now.

umesh-timalsina, 2022-01-14:
While building a pattern using webgme's regular expression is a valid solution, here's what I don't like about it:

Technically, we should be able to build a single regular expression for all possible units. Libraries like [pint](https://github.com/hgrecco/pint), [unyt](https://github.com/yt-project/unyt) and [js-quantities](https://github.com/gentooboontoo/js-quantities/blob/2c5442d5e351f499950aee3fd8dd17b97a598f4f/src/quantities/parse.js#L5-L23) do it in a simple and efficient manner. The downside of course is that we will not be using the ontology if we rely on this approach. A good solution would be us using a single regex/grammar for parsing units and scalar values without having to rely on bunch of attributes' regex pattern while also being able to specify this pattern via the ontology. One possibility is to incorporate the patterns as GME nodes in the language itself (which is also not ideal).

brollb, 2022-01-14:
I don't expect defining the regular expressions to be a common task (as it is in the ontology definition rather than in any part of the labeling workflow) and making regex easier is orthogonal to our goal so it probably isn't the best place to invest a lot of time. I would use some of the existing regex tools like https://regex101.com/, https://regexr.com/, or https://regex-generator.olafneumann.org/.