symbench/scm #381: Include PartNumber for non-existent circuit?

umesh-timalsina, 2021-11-12:
For the following circuit, part number exists (by the image title). Should we include while labeling the image? If so where? If not we will not be able to resolve its name.
[image]

brollb, 2021-11-12:
Ideally, we should record metadata (ie, filename) somewhere. Currently, we "save" it by preserving it for the annotations file but it isn't clear if this is the best way to record it.

In other words, as long as we can safely assume that unnamed "Circuit" annotations are the same as the part name title, it is safe to leave them as is. If we run into a diagram where it includes circuits where this isn't the case, then we will need to determine how the user should annotate the image to resolve any ambiguity.

umesh-timalsina, 2021-11-12:
Okay. I have added the annotations without a name