symbench/scm #387: Add support for references in the ontology

brollb, 2021-11-16:
Currently, `details` can only contain a list of text attributes. It would be great to update them to be able to include references as well. This will be a little involved as it will require updates to the widget and ontology generator as well.

Here is an initial draft of a possible representation in the ontology.json file:
```javascript
        "properties": [                                
            {"name": "value", "type": "string", "regex": ""},     // regex doesn't need to be supported in this PR, of course
            {"name": "source", "type": "reference"},              
            {"name": "destination", "type": "reference"}          
        ]
```

brollb, 2021-11-16:
Note:  I did rename "details" to "properties" here.

brollb, 2021-11-16:
Another consideration is how this will be functional within the widget. Namely, how will the annotation be presented to the user? The ID is not helpful; the component class is ambiguous.

The most straightforward seems to be something like showing the component class then, on hover, adding a class to the referenced annotation so it blinks or changes size or something.

umesh-timalsina, 2021-11-17:
What will be the purpose of these references? Are they going to be used for handling source and destination for abstract connections?

brollb, 2021-11-17:
Yep, exactly

umesh-timalsina, 2021-11-17:
>The most straightforward seems to be something like showing the component class then, on hover, adding a class to the referenced annotation so it blinks or changes size or something.

I think there should be two distinct types of workflows:
1. Since, most of the times we will be handling abstract connections that overlap certain annotations, we are going to create a SVG overlay in the viewer, that draws a wire to the overlapping bounding boxes. This is inline with how we predict ports.
2. This workflow can be manual as well (This is tangentially related to discontinuous connections that are labeled the same name), where in you can essentially extend a wire(via svg overlay) that can manually draw a wire from source to destinations. The visibility can be hidden by default and only shown when user is intentionally asking to view them.

brollb, 2021-11-17:
> that draws a wire to the overlapping bounding boxes. This is inline with how we predict ports.

This is news to me. When was any decision made on how references should be shown to the users?

Have you been assuming we will need to add visual lines to the diagram so they are detected by the connection detection step? If so, I don't think that is necessary as these will need to be detected differently when automatically annotating new images.

umesh-timalsina, 2021-11-17:
>Namely, how will the annotation be presented to the user? The ID is not helpful; the component class is ambiguous.
> This is news to me. When was any decision made on how references should be shown to the users?

Wasn't this the question you posed?

>Have you been assuming we will need to add visual lines to the diagram so they are detected by the connection detection step? If so, I don't think that is necessary as these will need to be detected differently when automatically annotating new images.

When you think of drawing connections, I don't think it intuitively makes sense to connect two elements just by clicking on the boxes.

umesh-timalsina, 2021-11-17:
> Have you been assuming we will need to add visual lines to the diagram so they are detected by the connection detection step? If so, I don't think that is necessary as these will need to be detected differently when automatically annotating new images.

Here's what I have been thinking: 
1. An optional visual queue in the labeling tool, that shows `source` and `destination` for Abstract connection
2. On a related note this can also be used to identify Connections that are not actually connected (see the "C" pseudo-node below):https://github.com/symbench/scm/issues/269#issuecomment-933712412. This part will be tricky because we don't know we that we want to draw annotations for the disconnected ends and connect them by adding one more category `continued`.

brollb, 2021-11-17:
> Wasn't this the question you posed?

Yeah, I posed the question but was surprised when you responded as if it had already been decided (referring to "we are going to create a SVG overlay in the viewer that draws a wire to the overlapping bounding boxes"). It sounds like you meant "we could create an SVG overlay in the viewer that draws a wire to the overlapping bounding boxes". Is that what you meant?

umesh-timalsina, 2021-11-17:
> I think there should be two distinct types of workflows:

I prefaced this with I think, which is arguing semantics at this point. But that's my suggestion on how we should proceed.

commit 5b95fee5 (2021-11-18): Add references to ontology. Closes #387 (#391)
closed 2021-11-18 by brollb