symbench/scm #400: New Diagram Annotation UX

brollb, 2021-11-22:
The UX for annotating new circuit diagrams needs a little work. There are a few different approaches that could be taken here:
- a CLI that spins up the required servers from the given directory and serves an index of the images in the directory
- an "upload" feature from the browser where the user can upload individual files or a zip file

Down the road, it would be nice for the flow to fit nicely into the recognition pipeline so the CLI tools could provide image paths or URLs to open the annotations in the browser for review/editing.

umesh-timalsina, 2021-11-23:
A possible idea we have discussed as far as it enabling rapid prototyping is the following:
> I think we should probably just put together something simple for now but keep in mind how the UX should work down the road. For now, I think adding upload capabilities and posting the images to the recognition server when they are first opened, should probably work (suggested by @brollb).

umesh-timalsina, 2021-11-23:
Another possibility albeit more work is the following:

> Can we leverage something like rollup-plugin-copy when starting the sirv server? I was thinking that may be sirv could be levaraged for user uploaded data without sending anything to server (ofcourse any image processing for template matching requires images to be uploaded to the server). (suggested by @umesh-timalsina)

umesh-timalsina, 2021-11-23:
Discussions
> I gave the statelessness further thought and we will loose it if anything is handled from the python server.

> It would be nice to have it be stateless and integrate well with other tools/workflows. I am hoping it will make integration into the recognition pipeline smoother. That said, it might be nice to just add the bare minimum (ie, upload to the webpage and add an endpoint to the recognition server to manually add images to the cache) so Peter can give it a try sooner

umesh-timalsina, 2021-11-23:
> A possible idea we have discussed as far as it enabling rapid prototyping is the following:

Okay. I think a lot can be fleshed out mid development. I am starting with your suggestion and will report any issues I face. Here's what I plan to accomplish.
1. Show modal for upload if the  no images can be loaded in the viewer.
2. Upload the content from the client (prefarably using a socket interface). Write logic to customize annotation URLs if needed. This will save the images and annotations in the server's tempdir with a guid directory for the current session.
4. Add end-points to get uploaded images and annotations from the server. This can be combined with step 2 and same logic for displaying annotations will work.
5. Add download options for the current progress made.