symbench/scm #428: Add manual refresh button for attribute OCR

brollb, 2022-01-07:


umesh-timalsina, 2022-01-13:
Where should the button live?

umesh-timalsina, 2022-01-13:
Here's a proposed set of changes and a discussion is necessary around these changes:
1. Change the input field used on the `ComponentLabelWidget.svelte` to use `@smui/text-filed`. The reason is because we already have smui integration and the [text fields](https://sveltematerialui.com/demo/textfield/) already achieve what we want. They have custom icons helper texts and much more.
2. Where should the knowledge about this refresh live? Should we change the ontology for minor changes like this, if not it needs to be hardcoded. I was thinking we could have an input field like the following.
[image]
3. The material UI theme scss files now need to live on the top level not on the `Components/TitleBar`.

brollb, 2022-01-13:
I think it looks good. I don't think we need any additional info in the ontology since it is really only needed for the annotations that already use OCR for their fields.

[image]<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F11476842%2F149389060-1962e347-def7-4a59-9461-8a337dfe7092.png&data=04%7C01%7Cbrian.broll%40vanderbilt.edu%7Cc82e4f50ca904e9c2ec208d9d6c3ab7e%7Cba5a7f39e3be4ab3b45067fa80faecad%7C0%7C0%7C637776958146377678%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=pGSb0HyeXovhhl3gZeO1CbMD8hk7dzi5tFO3qIti1E8%3D&reserved=0>

umesh-timalsina, 2022-01-13:
I agree, but there are cases where a fields can have value and unit. Adding a refresh button just to the `value` should work in that case but is it the most intuitive?

brollb, 2022-01-13:
Good point. We can't really refresh just one of the fields... We could add a refresh button to both. On refresh, we could run OCR then assign the text to the different fields (based on regex from the ontology). Whichever button was refreshed could simply set the field accordingly and the other field (ie, value/unit) could be discarded.

It might be nice to first implement the first part of that so we can enter something like "10 k\Omega" in the value field and it could automatically move the units section to the unit field. The only problem with this would be the example Peter gave a while back about `1\Omega5` for `1.5 \Omega`.

umesh-timalsina, 2022-01-13:
>It might be nice to first implement the first part of that so we can enter something like "10 k\Omega" in the value field and it could automatically move the units section to the unit field. The only problem with this would be the example Peter gave a while back about 1\Omega5 for 1.5 \Omega.

I am not too much worried about the later part as this can be achieved incrementally (e.g. 1\Omega5 or 1.5 \Omega) by having regex support for different use cases. However, if we are to do the first part, I think this should be done separately i.e. #362.

> Good point. We can't really refresh just one of the fields... We could add a refresh button to both. On refresh, we could run OCR then assign the text to the different fields (based on regex from [...]

I think discarding could be an option. For example- Let's say you would do a snippet OCR and get the following value for resistance: `5K`. Ideally, we should be able to split this into two parts 5 for value and `K\Omega` for unit, regardless if the underlying bounding box doesn't change, I think we will end up with the same results. So refreshing `unit` should just change the `unit` field and refreshing `value` should change the `value` field (but results will seem unchanged).